openblockcc/openblock-desktop #1: 我正在尝试在Linux平台构建它，但是出错了。

szqjslh, 2021-06-02:
因为Windows系统难以维护，打算使用ubuntu/debian/deepin之类的系统当构建环境和运行环境。

可重现的代码:

仓库: https://github.com/openblockcc/openblock-desktop/
commit id: v2.1.0

可以重现的环境:

dockerfile:
```
FROM ubuntu:latest

ENV DEBIAN_FRONTEND=noninteractive
RUN apt update -y && \
apt-get install npm git iproute2 -y && \
apt-get install libasound2 libxss1 libgtk-3-0 libgdk-pixbuf2.0-0 libnss3 -y && \
npm install -g node-gyp --registry=https://registry.npm.taobao.org
```

运行 npm run dist 的时候遇到了以下错误:
```
> openblock-desktop@2.1.0 doBuild /project/openblock-desktop
> node ./scripts/electron-builder-wrapper.js "--mode=dist"

/project/openblock-desktop/scripts/electron-builder-wrapper.js:148
        throw new Error(`Could not determine targets for platform: ${process.platform}`);
        ^

Error: Could not determine targets for platform: linux
    at calculateTargets (/project/openblock-desktop/scripts/electron-builder-wrapper.js:148:15)
    at main (/project/openblock-desktop/scripts/electron-builder-wrapper.js:196:29)
    at Object.<anonymous> (/project/openblock-desktop/scripts/electron-builder-wrapper.js:203:1)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! openblock-desktop@2.1.0 doBuild: `node ./scripts/electron-builder-wrapper.js "--mode=dist"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the openblock-desktop@2.1.0 doBuild script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/openblockbuilder/.npm/_logs/2021-06-02T02_02_01_177Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! openblock-desktop@2.1.0 dist: `npm run clean && npm run compile && npm run fetch:all && npm run doBuild -- --mode=dist`
npm ERR! Exit status 1
```

我查阅了报错脚本中的代码，它是有linux平台的:

```
/**
 * @returns {string} - an `electron-builder` flag to build for the current platform, based on `process.platform`.
 */
const getPlatformFlag = function () {
    switch (process.platform) {
    case 'win32': return '--windows';
    case 'darwin': return '--macos';
    case 'linux': return '--linux';
    }
```

实在没搞懂为啥它会失败。

zhengyangliu, 2021-06-03:
目前 Openblock 还没有彻底适配 Linux , electron-builder.yam 没有面向 Linux 的构建目标。
https://github.com/openblockcc/openblock-desktop/blob/00417199d01a96ff0c222a504b410bb21e7525de/electron-builder.yaml#L16-L56
且项目使用的 Tools 二进制工具也没有打包 Linux 版本的所以还是等等吧。在近期如果 v2.1.1 没有什么大的问题的话我会对 mac 以及 linux 做适配。

closed 2021-06-03 by zhengyangliu

szqjslh, 2021-06-09:
好的，期待Linux 版本的到来