openblockcc/openblock-desktop #3: Is possible to add AI Extension.?

ircbloqcc, 2021-06-28:
Hi, 
First, i will say thanks for making this openblock source.I am looking for this open source for a long time.
Now i am trying to integrate the Google's Teachable Machine to this source. But i can't able do that.

lf possible, Can you able to add AI extension in this source?

Thanks in Advance.

zhengyangliu, 2021-06-29:
Hi,
I'm not very familiar with Google's Teachable Machine. Can you give me a general idea of how you plan to use it.

ircbloqcc, 2021-06-29:
Hi,
I just want to interface with Arduino with Teachable machine Trained data over serial communication or bluetooth.
I have tried to add that feature using https://github.com/mitmedialab/prg-extension-boilerplate from this. 
But i can't able to complete it. Since i new to this programming, i can't figure it out. 
Hope you will get that what i trying to do.

zhengyangliu, 2021-06-29:
I'll see if I can do that. But if it's too complex to be included in the core development plan, we won't be able to achieve it in a short time.

closed 2021-07-24 by zhengyangliu