openblockcc/openblock-desktop #6: Macos run dist error

ekem57, 2021-07-10:
Hello there,
I want to add my own robot to openblock-desktop and package it. I keep getting the same error on my computer macbook operating system catalina 10.15 npm run dist partition.

Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future

> fsevents@1.2.13 install /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/fsevents
> node install.js

  SOLINK_MODULE(target) Release/.node
  CXX(target) Release/obj.target/fse/fsevents.o
  SOLINK_MODULE(target) Release/fse.node

> @serialport/bindings@9.0.7 install /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> nodegit@0.27.0 install /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/nodegit
> node lifecycleScripts/preinstall && node lifecycleScripts/install

[nodegit] Running pre-install script
[nodegit] Running install script
node-pre-gyp WARN Using request for node-pre-gyp https download
[nodegit] Success: "/Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/nodegit/build/Release/nodegit.node" is installed via remote
[nodegit] Completed installation successfully.

> @abandonware/noble@1.9.2-13 install /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/nothing.o
  LIBTOOL-STATIC Release/nothing.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: archive library: Release/nothing.a the table of contents is empty (no object file members in the library define global symbols)
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
  CXX(target) Release/obj.target/binding/lib/mac/src/napi_objc.o
  CXX(target) Release/obj.target/binding/lib/mac/src/ble_manager.o
  CXX(target) Release/obj.target/binding/lib/mac/src/objc_cpp.o
  CXX(target) Release/obj.target/binding/lib/mac/src/callbacks.o
  SOLINK_MODULE(target) ../lib/mac/native/binding.node
  COPY Release/binding.node
  SOLINK_MODULE(target) Release/noble.node

> core-js-pure@3.6.5 postinstall /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/core-js-pure
> node -e "try{require('./postinstall')}catch(e){}"

> core-js@3.8.2 postinstall /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/global-agent/node_modules/core-js
> node -e "try{require('./postinstall')}catch(e){}"

> nodegit@0.27.0 postinstall /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/nodegit
> node lifecycleScripts/postinstall

> electron@8.5.5 postinstall /Users/emre/Downloads/openblock-desktop-2.1.0/node_modules/electron
> node install.js

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@0.5.3-7 (node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-7: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

zhengyangliu, 2021-07-13:
This is because I used a npm package that is not supported under Mac. I will repair and release the OpenBlock desktop for Mac in a short time

zhengyangliu, 2021-07-21:
The mac verison is added in 15edc3edff1f588df1b7f47f0559b5bce35e3d20.

closed 2021-07-21 by zhengyangliu