openblockcc/openblock-desktop #7: error on : npm start

ffsher, 2021-07-12:
I study the link : https://openblockcc.gitee.io/wiki/zh/developer/develop-from-source/pack-desktop-software/

Every thing is fine til command: "npm run dist" ,when I followed steps detailed on openblock-desktop.

### Actual Behavior & log.
ffsher@DESKTOP-OAHCTK8:/mnt/c/Scratch_build/Openblock/openblock-desktop$ **sudo npm run dist**

> openblock-desktop@2.2.0 dist /mnt/c/Scratch_build/Openblock/openblock-desktop
> npm run clean && npm run compile && npm run fetch:all && npm run doBuild -- --mode=dist

> openblock-desktop@2.2.0 clean /mnt/c/Scratch_build/Openblock/openblock-desktop
> rimraf ./dist ./statics ./external-resources ./tools ./translations ./firmwares ./drivers

> openblock-desktop@2.2.0 compile /mnt/c/Scratch_build/Openblock/openblock-desktop
> mkdirp ./dist && electron-webpack --bail --display-error-details --env.minify=false --no-progress

[1398:0712/144014.697222:FATAL:electron_main_delegate.cc(211)] **Running as root without --no-sandbox is not supported**. See https://crbug.com/638180.
...........

###  I try

I reference : 
https://stackoverflow.com/questions/59203085/fatal-running-as-root-without-no-sandbox-is-not-supported-using-electron-7-1
And seems to **electron**.

As attached link, addition of parameter "**--no-sandbox**" is workaround.

But, which file/script i can put the parameter? 
Or, any other suggestion to pass the error

### Operating System and Browser
version 
Ubuntu 16.04.7 LTS
npm : 6.14.13
electron : v8.5.5
node : v14.17.2

thanks for you advise.

ircbloqcc, 2021-07-12:
In Desktop version V2.2.0

If i tried to open saved project, i getting error like this.
this problem is not in V2.1.1.
How to solve this?

[image]

zhengyangliu, 2021-07-13:
At present, I haven't studied the process of packaging software under Linux. However, it seems that this parameter should not appear. Have you ever tried to run the command without root permission.

ffsher, 2021-07-27:
Actually, i am running WSL and got some idea by the link :
https://stackoverflow.com/questions/63784533/cant-run-electron-on-win10-ubuntu-subsystem

WSL need to be switched "metadata" and it work

closed 2021-07-27 by zhengyangliu