openblockcc/openblock-desktop #8: Bug in Desktop application Version 2.2.0

ircbloqcc, 2021-07-12:
In Desktop Version 2.2.0.
If i tried to open saved project,i am getting error like this.
But in V2.1.1, this problem was not there..

How to Solve this Error?

[image]

zhengyangliu, 2021-07-13:
I've received similar bug reports, but I haven't been able to reproduce this situation stably, so it's hard to fix it. But I'll try to fix it in the next 2.2. X release.

ircbloqcc, 2021-07-13:
Thank you,
Expecting soon with AI extension

zhengyangliu, 2021-07-13:
At present, AI extension are not included in the mainline development plan. You can launch a vote to see which function of AI extensions people prefer, or you can develop one by yourself. I will try my best to help you.

ircbloqcc, 2021-07-13:
Yes, I will try to add. But still i am new in this programming. How much time it will take to me i dont know..
Thank you :)

zhengyangliu, 2021-07-19:
This bug should have been fixed In https://github.com/openblockcc/openblock-gui/commit/21b9b56c186d50bab4a8b0a948dc13e1d3441279.

closed 2021-07-19 by zhengyangliu

ircbloqcc, 2021-07-28:
Still the issues there in Version 2.2.2.
If i opened the Saved project from location where i saved by Double click. OpenBlock App is opening But Project is not Loading. It shows EMPTY.
[image]