openblockcc/openblock-desktop #10: After Resetting esp, data not receive in serial monitor.

ircbloqcc, 2021-07-19:
In my esp8266, i tried to send data from esp to computer. Without pressing reset, everything working fine. But if i press reset button, serial monitor not receiving any data.

How to solve this issue?

zhengyangliu, 2021-07-19:
This is related to the status of DTR and RTs of the serial port. There should be a level that affects the startup of esp after reset, which is related to the structure of the hardware design. At present, you can disconnect the device and use the serial assistant to debug it. I'll see how to fix it later.

zhengyangliu, 2021-07-28:
This bug has been fixed in https://github.com/openblockcc/openblock-desktop/releases/tag/v2.2.2

closed 2021-07-28 by zhengyangliu