openblockcc/openblock-desktop #12: "out of memory" error occurred during development using esp32.

hhs4420, 2021-07-27:
_Hello, I am developing a block using esp32._
_All other codes were fine, but an error called "out of memory" occurred during upload when creating blocks using the ArduinoIotCloud.h header file and the Arduino_ConnectionHandler.h header file._
_Below is the code I wrote._

#include <Arduino.h>
#include <ArduinoIoTCloud.h>
#include <Arduino_ConnectionHandler.h>
#include <DHT.h>

const char SSID[] = "WIFIID";
const char PASS[] = "WIFIPASSWORD";
const char DEVICE_KEY[] = "DEVICEKEY";
const char THING_ID[] = "THINGID";
const char DEVICE_LOGIN_NAME[] = "DEVICELOGINNAME";

WiFiConnectionHandler ArduinoIoTPreferredConnection(SSID, PASS);

bool led1;
int led2;
float temp;
float humi;
DHT dht_19(19, DHT11);

void setup() {
  ArduinoCloud.setSecretDeviceKey(DEVICE_KEY);
  ArduinoCloud.setThingId(THING_ID);
  ArduinoCloud.setBoardId(DEVICE_LOGIN_NAME);
  ArduinoCloud.begin(ArduinoIoTPreferredConnection);
  ArduinoCloud.addProperty(led1, READWRITE, ON_CHANGE, NULL);
  ArduinoCloud.addProperty(led2, READWRITE, ON_CHANGE, NULL);
  ArduinoCloud.addProperty(temp, READWRITE, ON_CHANGE, NULL);
  ArduinoCloud.addProperty(humi, READWRITE, ON_CHANGE, NULL);
  pinMode(15, OUTPUT);
  ledcSetup(0, 2000, 8);
  ledcAttachPin(32, 0);
  dht_19.begin();
}

void loop() {
  ArduinoCloud.update();
  digitalWrite(15, led1);
  ledcWrite(0, led2);
  temp = dht_19.readTemperature();
  humi = dht_19.readHumidity();
}

_This is the picture that caused the problem. There is a problem uploading as shown below._
[image]
[image]

_I tried it on another computer, but the same problem occurs. Is there a way to solve it?_

zhengyangliu, 2021-07-27:
The second picture can't be displayed. Can this code be used in the Arduino ide? If not, I think you should ask this question under the esp32 project.

hhs4420, 2021-08-05:
> The second picture can't be displayed. Can this code be used in the Arduino ide? If not, I think you should ask this question under the esp32 project.

This code works normally on Arduino ide.
The second picture shows the error code "Out of Memory" when checked on the web through openblock-gui.