openblockcc/openblock-desktop #13: Arduino board cann't be found in Openblock

ffsher, 2021-07-27:
### Actual Behavior
In Openblock software,  it shows information :  
                                         1. Make sure you have OpenBlock Link installed and running
                                         2. check that USB cable is connected or BT is enable

### Steps to Reproduce
 1. followed link : https://openblockcc.gitee.io/wiki/zh/developer/develop-from-source/quick-start/
     "clone & install"   
      openblock-link / openblock-resource / openblock-gui are running

 2. inserting Arduino into USB hub. Arduino board is already upload "firmata" by arduino software.
 
 3 . open Openblock software,  I click "No device selected" on top bar.

zhengyangliu, 2021-07-27:
Try to select the show all connected devices checkbox. This turns off the pid/vid filter and displays the control board using the third-party usb2serial chip.
[image]

zhengyangliu, 2021-07-27:
This indicates that the GUI cannot connect to the link port. I think you need to check whether link is really running.