openblockcc/openblock-desktop #14: Project is not Loading

ircbloqcc, 2021-07-28:
Still the issues there in Version 2.2.2.
If i opened the Saved project from location where i saved by Double click. OpenBlock App is opening But Project is not Loading. It shows EMPTY.
[image]

_Originally posted by @ircbloqcc in https://github.com/openblockcc/openblock-desktop/issues/8#issuecomment-888316053_

zhengyangliu, 2021-07-28:
Pelese upload your project file. I need it to check the bug.

ircbloqcc, 2021-07-28:
If i Load from the APP, it is loading. But it is not Loading when i double click the Project file directly from saved Location.
This is the project file.
[OpenBlock Project.zip](https://github.com/openblockcc/openblock-desktop/files/6893713/OpenBlock.Project.zip)

this is the screenshot that i load from application.

[image]

ircbloqcc, 2021-07-28:
[image]

[image]

zhengyangliu, 2021-07-28:
Indeed, the bug still exists, but I don't know what's wrong. It is expected to take a lot of time to fix it. At present, it is only recommended not to use this function.

ircbloqcc, 2021-07-28:
Yeah.. thank you.. 
But this bug is not there in V2.1.1

zhengyangliu, 2021-07-28:
This bug should be caused by some interfaces not updated after merging the official content of LLK in v2.2.x

ircbloqcc, 2021-07-28:
I think the problem is loading with device id. 
Without selecting device. if i saved the project, it is opening the project without error.

ircbloqcc, 2021-07-31:
Found the bug. It is in openblock-vm. due to this [clearMonitor function]( https://github.com/openblockcc/openblock-vm/commit/e8d518c5d7244e40c8220e87f6f3a4e7fcb084b3#diff-9362558266dae3ef23277f5983150d157e3e325843cdb30ce0b0b64106337574)

zhengyangliu, 2021-07-31:
Ah ha. This seems very likely to be the cause of the problem. Have you verified this conclusion? It's too late today. I'll test it tomorrow. If so, I'll find a way to fix it

ircbloqcc, 2021-07-31:
Yeah I have tested by commenting that option. Its worked.

zhengyangliu, 2021-08-02:
This bug has been fixed in https://github.com/openblockcc/openblock-vm/commit/8f76a3e8fe72689d8cfd7e57ecfaa6c4511f9188.

zhengyangliu, 2021-08-02:
> This bug has been fixed in [openblockcc/openblock-vm@8f76a3e](https://github.com/openblockcc/openblock-vm/commit/8f76a3e8fe72689d8cfd7e57ecfaa6c4511f9188).

The repaired code will be used in the v2.2.3 packaging. I will close this issue if the test is no problem.

ircbloqcc, 2021-08-02:
Tested.. Its Working..
Thank you..

zhengyangliu, 2021-08-03:
fixed in https://github.com/openblockcc/openblock-desktop/tree/v2.2.3

closed 2021-08-03 by zhengyangliu