openblockcc/openblock-desktop #15: Bug: adding comments to Device extensions

ircbloqcc, 2021-08-02:
After i add comment for device extension block. If i try to load the project, there is another comment window appear which cannot able to delete also.

[image]

zhengyangliu, 2021-08-02:
Can you upload your project file? I can't reproduce this bug.

ircbloqcc, 2021-08-02:
If i select device and add extension, then Its not coming.

Project file:
[OpenBlock Project.zip](https://github.com/openblockcc/openblock-desktop/files/6914385/OpenBlock.Project.zip)

[image]

[image]

ircbloqcc, 2021-08-02:
https://user-images.githubusercontent.com/86342573/127805214-83999a2e-dbc8-4a65-98da-62735d071675.mp4

https://user-images.githubusercontent.com/86342573/127805237-9dead7e0-fc15-4886-bcb0-fbb6190a7fa2.mp4

ircbloqcc, 2021-08-02:
[image]

[image]

ircbloqcc, 2021-08-02:
https://user-images.githubusercontent.com/86342573/127806458-194020cb-e281-42cc-8930-8b8e9a5c7fdd.mp4

https://user-images.githubusercontent.com/86342573/127806491-b0a3d663-605a-4024-8c55-4159aec6afc5.mp4

ircbloqcc, 2021-08-02:
[image]

[image]

ircbloqcc, 2021-08-02:
[image]

[image]

ircbloqcc, 2021-08-02:
[image]

[image]

zhengyangliu, 2021-08-02:
This bug has been fixed in https://github.com/openblockcc/openblock-vm/commit/46d8caaa181037a5e5cdd18d6f928586cdfe80e4

zhengyangliu, 2021-08-02:
> This bug has been fixed in [openblockcc/openblock-vm@46d8caa](https://github.com/openblockcc/openblock-vm/commit/46d8caaa181037a5e5cdd18d6f928586cdfe80e4)

The repaired code will be used in the v2.2.3 packaging. I will close this issue if the test is no problem.

ircbloqcc, 2021-08-02:
Tested.. Its Working..
Thank you..

zhengyangliu, 2021-08-03:
fix in https://github.com/openblockcc/openblock-desktop/tree/v2.2.3

closed 2021-08-03 by zhengyangliu