openblockcc/openblock-desktop #18: Bug: After loading the saved project, i cannot able to edit the text.

ircbloqcc, 2021-08-17:
If i try to edit the loaded project in openblock, i cannot able to edit text. Even after try to open a NEW project also having same problem

[image]

Actually text selection should show blue color. But it shows in gray color. Keyboard are not working. If i save the loaded project again, then only it can able to edit the text.

ircbloqcc, 2021-08-17:
[image]

zhengyangliu, 2021-08-28:
I also found this bug and am looking for what went wrong。

zhengyangliu, 2021-08-28:
I don't know which commit fixed the bug. But this bug no longer exists. I think it may be caused by the infinite reentry of the startheartbeat function in the previous code.  https://github.com/openblockcc/openblock-vm/commit/63aeb45597d520eeda398c3cc6c94c26282e5454

zhengyangliu, 2021-08-28:
This bug will be fixed in v2.2.5

zhengyangliu, 2021-08-28:
The bug sitll exists.

ircbloqcc, 2021-08-28:
:( But this problem also there in Official Scratch desktop application.

zhengyangliu, 2021-08-28:
WTF. I found the operation to remove the error: move the focus out of the electron and come back. But I don't know how to fix this bug。

zhengyangliu, 2021-08-28:
This is a bug from electron. https://github.com/electron/electron/issues/19977. We need to replace all alert and comfim to fix this problem.

zhengyangliu, 2021-08-29:
After i replace all alert and comfim to electron dialog. This bug has been fixed.

closed 2021-08-29 by zhengyangliu