openblockcc/openblock-desktop #20: Release Update checking in ob desktop version.

ircbloqcc, 2021-08-28:
is it possible to add for checking package update option in OpenBlock?
Because each time, we need to download the package from GitHub releases.
If you add check for update option for full package instead of resources, it would be notice instantly. Also very easy to update the package.

zhengyangliu, 2021-10-21:
It's not hard to add this feature, but for Chinese users, updating through github is too slow, and I've been trying to figure out how to fix it, so I haven't done it yet. Maybe in the near future I'll rent a server to fix this problem.

zhengyangliu, 2022-03-06:
This feature has been supported in v2.2.3 https://github.com/openblockcc/openblock-desktop/releases/tag/v2.3.0.

closed 2022-03-06 by zhengyangliu