openblockcc/openblock-desktop #21: Bug: I cant able to load project

ircbloqcc, 2021-09-08:
Still some issues in loading project. I cant load the project in OB 2.2.6

[image]

zhengyangliu, 2021-09-08:
Can you show me your project file? the index.js in the pack.------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年9月8日(星期三) 中午12:43
***@***.***&gt;;
***@***.***&gt;;
主题:&nbsp;[openblockcc/openblock-desktop] Bug: I cant able to load project (#21)

ircbloqcc, 2021-09-08:
this is the project file

[asdas.zip](https://github.com/openblockcc/openblock-desktop/files/7126313/asdas.zip)

zhengyangliu, 2021-09-16:
Is this file is  created in the old version？ Its internal deviceextensions configuration is empty, which is due to a bug in the previous version. If you create with a new version, this problem should not exist.

[image]

zhengyangliu, 2021-09-16:
After you confirm, I will close this issue.

ircbloqcc, 2021-09-16:
> Is this file is created in the old version？ Its internal deviceextensions configuration is empty, which is due to a bug in the previous version. If you create with a new version, this problem should not exist.
> 
> [image]

No i have created with new openblock v2.2.6 only. For conformation vm prerelease is latest only. 

Also i created new and checked again..

[image]

zhengyangliu, 2021-09-16:
I misjudged the problem. Can you show me how to create this bug causing project?

ircbloqcc, 2021-09-16:
[image]

zhengyangliu, 2021-09-16:
It's a tough problem. I'll see what's wrong。

ircbloqcc, 2021-09-16:
I think, problem is with vm only. its not clear the old target, if i edit a old project and saved again means its not clearing old targets. instead of that its adding the stage in side the json file.

[image]

zhengyangliu, 2021-09-16:
I need to see where the bug is caused.

zhengyangliu, 2021-09-18:
This bug has been solved in https://github.com/openblockcc/openblock-desktop/commit/17315fd85d6198fcde48ff846becea827ea58692.

ircbloqcc, 2021-09-18:
Sorry to say, still some issues..
if try to load same project again, code generation area is blank. if i click the costume tab and get back, then code is generating. 
if i load same project, com port is not disconnecting. also upload button not responding. 

[image]

[image]

zhengyangliu, 2021-09-18:
I can't repeat the first question. I will fix the problem of device disconnection in the next version. At present, you can manually fix this bug by unplugging the USB cable and reconnect the device.

zhengyangliu, 2021-09-18:
Can you show me how to trigger the first question？

ircbloqcc, 2021-09-18:
This is the problem

[image]

zhengyangliu, 2021-09-18:
When no deivce extenison is loaded, repeat load this project, the code area will be empty.

ircbloqcc, 2021-09-18:
Correct.. If any device extensions is loaded, code is generating,,
When no device extension is loaded, code area will be empty.

zhengyangliu, 2021-09-18:
Fortunately, the bug is not urgent this time, I can fix it a bit slower and relax..

ircbloqcc, 2021-09-18:
[image]

ircbloqcc, 2021-09-18:
> Fortunately, the bug is not urgent this time, I can fix it a bit slower and relax..

cool..

CeemaCR, 2021-09-29:
Hi Arthur, Hope you will release the next version soon. Waiting eagerly for a fix of this one....Thanks a lot.

zhengyangliu, 2021-09-29:
ok, I will update in is week.

CeemaCR, 2021-09-29:
Thank you so much...

CeemaCR, 2021-10-04:
Hi Arthur, hope the same change is applicable to the Mac version too

zhengyangliu, 2021-10-04:
Yes, they are updated synchronously, and they all use the same code.

CeemaCR, 2021-10-04:
okay thank you . Hope you are working on the Issue with the upload.  And the latest version .7 has given a hint to a lot of new upcoming updates like driver and all...Eagerly waiting to see your updates.

zhengyangliu, 2021-10-05:
Those bug has been fixed in https://github.com/openblockcc/openblock-desktop/releases/tag/v2.2.8.

zhengyangliu, 2021-10-05:
I will close this issue after you test the feedback。

CeemaCR, 2021-10-05:
Thank you so much. Will check and let you know

CeemaCR, 2021-10-05:
Seems like there is some issue. Can not see any other device other than Arduino  Uno

[image]

Checked set up file too,  when Arduino Nano is selected in set up file,  app got closed.

zhengyangliu, 2021-10-06:
The problem was fixed, and I re-released v2.2.8. This bug originated when I updated openblock-link when fixing the problem that microbit v2 could not download the firmware, and some npm packages that openblock-link relied on were updated. These updated packages will cause link crashes.

CeemaCR, 2021-10-06:
So, will there be a new release for the fix?

zhengyangliu, 2021-10-06:
Just re-download v2.2.8 directly.

CeemaCR, 2021-10-07:
Uno only is displayed and that too is not detecting. Same old issue

CeemaCR, 2021-10-07:
Me only having this issue or anyone else is facing this issue? Any suggestion?

CeemaCR, 2021-10-07:
> My mistake. First uninstall the software completely, and then use the
> clear cache function in the upper right corner of the newly installed
> software. Don't get the order wrong.

Thanks, you Arthur, will check it now..

CeemaCR, 2021-10-09:
Not sure, whether I should continue here or open a new issue. If I should open a new one, please let me know

As in the video, when we try to upload the code from .ob file, the device is getting disconneted. Any reason for that? And some times the upload takes a lot of time (not .ob file, the direct upload itself) , but, not always

https://user-images.githubusercontent.com/82566442/136644891-87ff7390-2030-45d8-837a-007bff05ef3c.mp4

zhengyangliu, 2021-10-09:
This is a usage problem. Just click this button when you want to connect the device. If you reload the project, it will automatically cut off the connection with the old device,
[image]
Click this button to connect the device. I think your previous usage may be a bit misunderstanding. haha.

cpyindrajith, 2021-10-09:
https://user-images.githubusercontent.com/81285660/136646807-02d2c7f3-169c-4ee5-b334-1e7f15ed9d39.mp4

cpyindrajith, 2021-10-09:
Showing error alert "Please connect a peripheral first" when trying to upload code from local file. I already connected Arduino UNO device with Openblock

ircbloqcc, 2021-10-09:
> Showing error alert "Please connect a peripheral first" when trying to upload code from local file. I already connected Arduino UNO device with Openblock

That is not error... Device has been disconnected.. You can reconnect then give upload...

zhengyangliu, 2021-10-09:
> > Showing error alert "Please connect a peripheral first" when trying to upload code from local file. I already connected Arduino UNO device with Openblock
> 
> That is not error... Device has been disconnected.. You can reconnect then give upload...
> 
> How many times you trying to load a file, device will get disconnect.. So each time you have to connect the com port and give upload..

Right answer.

cpyindrajith, 2021-10-09:
Do I need to reconnect the device every time I try to upload a saved file?