openblockcc/openblock-desktop #23: Not able to Upload the .ob file which is already saved

Ceema1973, 2021-09-17:
https://user-images.githubusercontent.com/82566442/133768068-ac7cdcc4-f97c-407b-99d1-466b51f18617.mp4

1. Created an .ob file and tried to save it on the computer
2. It's getting saved
3. Tried to load the saved project file. It successfully loaded. But, then is not able to upload it again.

Please see the video

https://user-images.githubusercontent.com/82566442/133768711-a0933054-23ca-4244-8a98-33ee56e8acb5.mp4

Ceema1973, 2021-09-17:
Saw that this issue is reported already. But, I am not getting any eeror

zhengyangliu, 2021-09-18:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        
            
        
    
    replied to you on gitter.Some time ago, my grandfather fell down and was hospitalized. I have been accompanying him in the hospital. Recently, I have been troubled by this bug for a long time. I have been concentrating on repairing this problem, so I didn't pay attention to my last email.

zhengyangliu, 2021-09-18:
Repeat with https://github.com/openblockcc/openblock-desktop/issues/21

closed 2021-09-18 by zhengyangliu

CeemaCR, 2021-09-18:
Hi Arthur,

    Thank you so much. Will implement this. Thanks a lot..

CeemaCR, 2021-09-20:
.ob code loading  is working fine now , but the upload button is not working.

zhengyangliu, 2021-09-20:
Yse, this bug has been reported by ircbloqcc. You can check this https://github.com/openblockcc/openblock-desktop/issues/21  to see how the bug is triggered and how to temporarily solve the problem. I'll fix it in the next version.