openblockcc/openblock-desktop #36: 当尝试更改变量名称时发生错误

Charles-Tang, 2021-12-23:
# 当尝试更改变量名称时发生错误

当我尝试更改变量名称时,我不小心将名称与另一个变量设成一样,软件提示已经有这个变量了,当我重新尝试修改的时候,无法在输入框中输入任何内容,创建变量的窗口也无法输入任何内容

zhengyangliu, 2022-01-31:
确实有这个问题，这是由于electron在win10后系统的一个bug导致，出现这个问题后，你可以将光标移动到软件外部点击聚焦其他内容，然后在回到软件中重新输入即可，不过同时我们会修复这个问题的。

zhengyangliu, 2022-03-08:
These problems all stem from the same bug: https://github.com/electron/electron/issues/20400#issuecomment-539586029. Under the windows system, the render thread inside the electron cannot use the altert popup window. We've fixed some of this before: https://github.com/openblockcc/openblock-desktop/commit/dcf119b4c27c8f85e810d3a13882af40570898c3, but didn't notice that alerts are also used in blocks, Now we need to put the alert from blocks to vm->gui->electron to fix this bug.

commit cfc68ddd (2023-04-01): Resolve #36
closed 2024-06-21 by zhengyangliu