openblockcc/openblock-desktop #40: Bug: Variable adding with custom blocks

ircbloqcc, 2022-01-04:
While adding set variable blocks in custom block, global keyword should add. but in OB is not happening. so it gives error.
### Expected Behavior

def on_button_a():
      global test
      test += 5

def on_pin0():
     global test
     test += 20`

### Actual Behavior

[image]

i have tried with python code. that also gives error.

[image]

zhengyangliu, 2022-01-04:
You are right. The code should be like this:
```python
from microbit import *

test = 0

test = 0

def repeat():
  if button_a.is_pressed():
    on_button_a()

def on_button_a():
  global test
  test += 1

while True:
  repeat()

```

zhengyangliu, 2022-01-04:
I will fix this soon.

closed 2022-01-05 by zhengyangliu