openblockcc/openblock-desktop #45: select different datatype in variable creation?

ircbloqcc, 2022-02-06:
Instead of creating default float datatype in variable.
Can you make an option to select data types like integer/float/string in Variable creation?

[image] [image]

zhengyangliu, 2022-02-06:
An overly complex variable system will cause confusion for beginners, and this function requires a lot of code modification. We are currently planning to write an advanced variable extension to implement this function, similar to the current 'insert source code'.

ircbloqcc, 2022-02-06:
Yeah, Got it :)

closed 2022-02-06 by zhengyangliu