openblockcc/openblock-desktop #49: Is Bluetooth communication possible?

mmirann, 2022-02-22:
First of all, thank you for providing a great open source.
I checked that there was a Bluetooth communication code on the `openblock-link`, but when I actually operated it, the following error appeared when connecting the device like micro:bit.

```
socket server listend:  http://127.0.0.1:20111
new connection
(node:9188) UnhandledPromiseRejectionWarning: Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\eel8\Documents\GitHub\openblock-link\node_modules\@abandonware\bluetooth-hci-socket\lib\usb.js:91:11)
    at BluetoothHciSocket.bindRaw (C:\Users\eel8\Documents\GitHub\openblock-link\node_modules\@abandonware\bluetooth-hci-socket\lib\usb.js:47:8)
    at Hci.init (C:\Users\eel8\Documents\GitHub\openblock-link\node_modules\@abandonware\noble\lib\hci-socket\hci.js:120:18)
    at NobleBindings.init (C:\Users\eel8\Documents\GitHub\openblock-link\node_modules\@abandonware\noble\lib\hci-socket\bindings.js:93:13)
    at Noble.startScanning (C:\Users\eel8\Documents\GitHub\openblock-link\node_modules\@abandonware\noble\lib\noble.js:145:20)
    at BLESession.discover (C:\Users\eel8\Documents\GitHub\openblock-link\src\session\ble.js:86:15)
    at BLESession.didReceiveCall (C:\Users\eel8\Documents\GitHub\openblock-link\src\session\ble.js:28:18)
    at BLESession.didReceiveRequest (C:\Users\eel8\Documents\GitHub\openblock-link\src\session\session.js:86:14)
    at BLESession.didReceiveMessage (C:\Users\eel8\Documents\GitHub\openblock-link\src\session\session.js:70:22)
    at BLESession.onMessage (C:\Users\eel8\Documents\GitHub\openblock-link\src\session\session.js:41:14)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:9188) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:9188) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

When I looked up the error on Google, Noble document explaining that [support is limited to a fixed list of hardware](https://github.com/noble/node-bluetooth-hci-socket#compatible-bluetooth-40-usb-adapters). Is Bluetooth communication actually possible?

zhengyangliu, 2022-02-24:
The code of the bluetooth part comes from the project https://github.com/Micircle/scratch-node-link. Since openblock is not currently using bluetooth, we have not verified the correctness of this part of the code.

zhengyangliu, 2022-02-24:
But this part of the code should be correct, if you are using windows, follow this instruction to use a supported Bluetooth 4.0 USB Adapter's and replace the driver for your adapter.
https://github.com/noble/node-bluetooth-hci-socket#windows

mmirann, 2022-02-24:
Thank you for your answer! :)

closed 2022-02-24 by mmirann