openblockcc/openblock-desktop #61: Can't able modify external-resource.

ircbloqcc, 2022-03-12:
In previous version openblock, if i try to add some extension means it will work.
but in new version, its not opening. 
if this is the feature means how can i add third party extension? how to check that?

[image]

zhengyangliu, 2022-03-12:
Follow the config file format in the official extension. Just delete the sha256 to skip the checksum step. I'll update the wiki later detailing this.

zhengyangliu, 2022-03-12:
Like this
```
{
    "updater": {
        "default": {
            "provider": "github",
            "repo": "openblockcc/external-resources-v2"
        }
    }
}

```

ircbloqcc, 2022-03-12:
cool... working..

closed 2022-03-12 by ircbloqcc