openblockcc/openblock-desktop #62: Bug: While uploading code, libraries not copied to Projectpath

ircbloqcc, 2022-03-19:
If i tried to upload a code with external libraries, Its getting Failed. If i tried to find the Library in project path, Library is not copied.

[image]
[image]

ircbloqcc, 2022-04-01:
any update regarding this issue??

zhengyangliu, 2022-04-03:
WTF? I will check what happen.

zhengyangliu, 2022-04-03:
I found it. Since the new version of openblock desktop has adjusted the naming and added a space in the middle, the cache path contains space characters, which makes the arduino cli parsing error. I will fix this soon.

zhengyangliu, 2022-04-03:
[image]

zhengyangliu, 2022-04-03:
In v2.2.9 this problem should exist

zhengyangliu, 2022-04-04:
My mistake, not the cause of the problem.. This is because in the v2.3.0 version, a cli tool will be used to optimize the external resource when it is released, and useless files will be deleted to reduce the library size. But I deleted the library.properties file by mistake, which caused the arduino cli to not recognize the library correctly and cause this error. Just update external-reource and i will fix this soon.

zhengyangliu, 2022-04-04:
Fixed in https://github.com/openblockcc/openblock-resource-cli/commit/16bb06d1c931563657af123b15472fa3ea06d753.
The software will automatically detect external-resource v0.0.4 and update it to fix.

closed 2022-04-04 by zhengyangliu