openblockcc/openblock-desktop #70: 编译时报错

lensli, 2023-05-18:
git clone https://github.com/openblockcc/openblock-desktop
cd openblock-desktop
npm i  的时候<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
 1: 00007FF6283461BF napi_wrap+133519
 2: 00007FF6282DF6D6 DSA_meth_get_flags+63062
 3: 00007FF6282E056D node::OnFatalError+301
 4: 00007FF628BC599E v8::Isolate::ReportExternalAllocationLimitReached+94
 5: 00007FF628BAA78D v8::SharedArrayBuffer::Externalize+781
 6: 00007FF628A53F8C v8::internal::Heap::EphemeronKeyWriteBarrierFromCode+1516
 7: 00007FF628A5F3AA v8::internal::Heap::ProtectUnprotectedMemoryChunks+1258
 8: 00007FF628A5C4E9 v8::internal::Heap::PageFlagsAreConsistent+2457
 9: 00007FF628A51081 v8::internal::Heap::CollectGarbage+2049
10: 00007FF628A4F285 v8::internal::Heap::AllocateExternalBackingStore+1349
11: 00007FF628A6F6FB v8::internal::Factory::NewFillerObject+203
12: 00007FF62879DE31 v8::internal::interpreter::JumpTableTargetOffsets::iterator::operator=+1409
13: 00007FF628C50DCD v8::internal::SetupIsolateDelegate::SetupHeap+474253
14: 00007FF628C51C7D v8::internal::SetupIsolateDelegate::SetupHeap+478013
15: 000002AC97718B8A  请问怎么回事

lensli, 2023-05-18:
这个错误是 npm run dist 时报错的 安装好了已经

lensli, 2023-05-18:
进度卡在92% chunk asset optimization TerserPlugin
<--- Last few GCs --->

[4568:0000021B2CAA4E80]     8836 ms: Mark-sweep (reduce) 1835.0 (1845.8) -> 1834.8 (1848.5) MB, 285.6 / 0.0 ms  (+ 0.1 ms in 554 steps since start of marking, biggest step 0.0 ms, walltime since start of marking 435 ms) (average mu = 0.542, current mu = 0[4568:0000021B2CAA4E80]     9602 ms: Mark-sweep (reduce) 1835.8 (1841.5) -> 1835.8 (1842.3) MB, 764.9 / 0.0 ms  (average mu = 0.262, current mu = 0.002) allocation failure scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: MarkCompactCollector: young object promotion failed Allocation failed - JavaScript heap out of memory
 1: 00007FF72C7A61BF napi_wrap+133519
 2: 00007FF72C73F6D6 DSA_meth_get_flags+63062
 3: 00007FF72C74056D node::OnFatalError+301
 4: 00007FF72D02599E v8::Isolate::ReportExternalAllocationLimitReached+94
 5: 00007FF72D00A78D v8::SharedArrayBuffer::Externalize+781
 6: 00007FF72CEB3F8C v8::internal::Heap::EphemeronKeyWriteBarrierFromCode+1516
 7: 00007FF72CE9EA0B v8::internal::NativeContextInferrer::Infer+59739
 8: 00007FF72CE83CDF v8::internal::MarkingWorklists::SwitchToContextSlow+56975
 9: 00007FF72CE979CB v8::internal::NativeContextInferrer::Infer+31003
10: 00007FF72CE8EAAD v8::internal::MarkCompactCollector::EnsureSweepingCompleted+6285
11: 00007FF72CE96C1E v8::internal::NativeContextInferrer::Infer+27502
12: 00007FF72CE9AC6B v8::internal::NativeContextInferrer::Infer+43963
13: 00007FF72CEA4662 v8::internal::ItemParallelJob::Task::RunInternal+18
14: 00007FF72CEA45F1 v8::internal::ItemParallelJob::Run+641
15: 00007FF72CE77C53 v8::internal::MarkingWorklists::SwitchToContextSlow+7683
16: 00007FF72CE8EF5C v8::internal::MarkCompactCollector::EnsureSweepingCompleted+7484
17: 00007FF72CE8D794 v8::internal::MarkCompactCollector::EnsureSweepingCompleted+1396
18: 00007FF72CE8B308 v8::internal::MarkingWorklists::SwitchToContextSlow+87224
19: 00007FF72CEB9D71 v8::internal::Heap::LeftTrimFixedArray+929
20: 00007FF72CEBBE65 v8::internal::Heap::PageFlagsAreConsistent+789
21: 00007FF72CEB1081 v8::internal::Heap::CollectGarbage+2049
22: 00007FF72CEAF285 v8::internal::Heap::AllocateExternalBackingStore+1349
23: 00007FF72CECF6FB v8::internal::Factory::NewFillerObject+203
24: 00007FF72CBFDE31 v8::internal::interpreter::JumpTableTargetOffsets::iterator::operator=+1409
25: 00007FF72D0B0DCD v8::internal::SetupIsolateDelegate::SetupHeap+474253
26: 00007FF72D0B1C7D v8::internal::SetupIsolateDelegate::SetupHeap+478013
27: 000001ECB7ADB614

zhengyangliu, 2023-05-19:
Duplicate of #29

closed 2023-05-19 by zhengyangliu