openblockcc/openblock-desktop #73: Heap memory error

jaimin1234568, 2024-07-09:
### Expected Behavior

_Please describe what should happen_

### Actual Behavior

_Describe what actually happens_

### Steps to Reproduce

_Explain what someone needs to do in order to see what's described in *Actual behavior* above_

### Operating System and Browser

_e.g. Mac OS 10.11.6 Safari 10.0_

jaimin1234568, 2024-07-10:
[image]

zhengyangliu, 2024-07-10:
Duplicate of https://github.com/openblockcc/openblock-desktop/issues/29

closed 2024-07-10 by zhengyangliu

jaimin1234568, 2024-07-11:
sir tried this solution and works but my repo does not run the npm run dist build command and does not make exe for my repository
sir my repository size is around 14 GB in store in my system
my system uses window 11, node version 14.16.1, and npm version 6.14.12 and the RAM is 8 GB in my system

jaimin1234568, 2024-07-11:
[image]

zhengyangliu, 2024-07-11:
`npm run build:dist` not `npm run dist build`. Just check the https://github.com/openblockcc/openblock-desktop/blob/59b2948696386870c6afc32129ccc0d5f11252b7/package.json#L8-L31 These are the available commands.

jaimin1234568, 2024-07-12:
try ths scrpt but error heap memory

jaimin1234568, 2024-07-12:
try this script and package.json file