openblockcc/openblock-desktop #74: exe in sprite and backdrop show but click when not show in canvas area

jaimin1234568, 2024-07-16:
my open block desktop in make exe in the new version and in add some things. this Exe is made completely and works in all programs. exe in the open any program is run in sprite and backdrops show but any sprites and backdrop clicks and open do not show any sprites and backdrops in my exe. please solve issue and quick reply in this issue

zhengyangliu, 2024-07-16:
I don't understand your description. Is threre any details? Can you record a video or something?

zhengyangliu, 2024-07-16:
I think I understand what you mean: when you double-click .ob to open the project, OpenBlock cannot process the contents of the project file correctly. I will fix it as soon as possible.

jaimin1234568, 2024-07-16:
[image]

jaimin1234568, 2024-07-16:
not show any sprite and backdrop in this area but  show all sprites and backdrop

zhengyangliu, 2024-07-16:
1. Check if your project file has obtained the correct icon
[image]
2. Check whether the project file is opened with an older version of the software(Maybe you still have an older version of OpenBlock running on your computer. Check the version number in the software title bar to confirm.)