openblockcc/openblock-desktop #78: npm start时一直提示TypeError: Cannot read properties of undefined (reading 'getVersion')报错

1006076811, 2025-03-09:
现在install都完成了，但是在运行npm run start的时候报错
┏ Electron -------------------

Debugger listening on ws://127.0.0.1:5858/08bf0de6-06ef-4384-aa15-4f1dd5403b49
For help, see: https://nodejs.org/en/docs/inspector

┗ ----------------------------
┏ Electron -------------------

TypeError: Cannot read properties of undefined (reading 'getVersion')
at eval (webpack-internal:///./src/main/OpenblockDesktopTelemetry.js:36:66)
at Module../src/main/OpenblockDesktopTelemetry.js (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/main.js:8682:1)
at webpack_require (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/webpack:/webpack/bootstrap:771:1)
at fn (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/webpack:/webpack/bootstrap:76:1)
at Module.eval (webpack-internal:///./src/main/index.js:16:84)
at eval (webpack-internal:///./src/main/index.js:638:30)
at Module../src/main/index.js (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/main.js:8718:1)
at webpack_require (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/webpack:/webpack/bootstrap:771:1)
at fn (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/webpack:/webpack/bootstrap:76:1)
at Object.0 (/Users/ryanleeb/Develop/openblocks/openblock-desktop/dist/main/main.js:8742:18)

┗ ----------------------------

1006076811, 2025-03-09:
m1 macos系统

zhengyangliu, 2025-06-03:
我手上没有类似构架的设备来做验证复现，但单纯从代码的角度来说，这个现象在正常情况下是不应该出现的，getVersion 这个函数报错是在 https://github.com/openblockcc/openblock-desktop/blob/fad36dc79f3ddd87e47fe5f6b6886dfb749536dc/src/main/OpenblockDesktopTelemetry.js#L23 但这个app对象是直接从 electron 中获取的，不可能为null，所以你应当检查自己的 packages 是否已经被正确无缺的安装。