openblockcc/openblock-desktop #79: issue: Device extension is not available.

ircbloqcc, 2025-05-17:
I have downloaded latest version of OpenBlock V2.6.0.

After i chosen the Arduino uno, In program mode, device Extension is not showning

[image]

zhengyangliu, 2025-06-03:
Have you tried using the Clear Cache button in the upper right corner of the software?

zhengyangliu, 2025-06-05:
As expected, starting from 2.6.0, OpenBlock will no longer use the cached external-resource by default, but will directly use the content under the installation path.The external-resource in the cache is read only when the extension update is downloaded to the cache directory.

zhengyangliu, 2025-06-05:
> Now i understood the issue. its because of that i have installed ExternalResource previously and uninstalled that. So this happens i think.
> 
> edited: Yes, i removed that path from env. Now its working.. :)
> 
> [image]

Yes, that's the reason. I think I need to add an uninstall script to delete this environment variable during uninstallation.